cu-ecen-aeld/final-project-TejaswiniLakshminarayan #4: GPS sensor data acquisition and preliminary driver development

TejaswiniLakshminarayan, 2022-11-07:


Ganeshkm10, 2022-11-13:
[image]

Sensor tested on Arduino Uno, Latitude issues seen on Uno which is expected

Ganeshkm10, 2022-11-13:
output: fix is now '1'

[image]

Ganeshkm10, 2022-11-14:
Arduino: Code pushed in this commit : [refer commit id](https://github.com/cu-ecen-aeld/final-project-Ganeshkm10/commit/92e00ca963ce39931d32a4b160a7cfd51675ddf7)

Ganeshkm10, 2022-11-14:
GPS sensor userspace code for beaglebone: [refer commit id ](https://github.com/cu-ecen-aeld/final-project-Ganeshkm10/commit/db0fd9a0a7fda550723db67a6755498fac487db5)

Ganeshkm10, 2022-11-14:
Update: Uart Device driver design is being carried over to upcoming sprints.
Apart from that, all other verification related to the GPS sensor is completed and userspace implementation is started. 
Hence closing this issue.

closed 2022-11-14 by Ganeshkm10