cu-ecen-aeld/final-project-TejaswiniLakshminarayan #8: SP2: Analysis of available GPS libraries

prvbharadwaj1731, 2022-11-14:
DoD

- Evaluation of gpsd [Refer here](https://github.com/ukyg9e5r6k7gubiekd6/gpsd) and libgps.
- Determine between the above options to choose one.

Ganeshkm10, 2022-11-20:
Installed gpsd on beaglebone and UART 4 was enabled and connected with GPS sensor.
If we have to design custom uart. Using this deamon would be difficult to integrate. 

[image]

Ganeshkm10, 2022-11-20:
Evaluated libgps library (available here - [libgps](https://github.com/wdalmut/libgps)) with some modifications to obtain only latitude and longitude values from the sensor. 

[image]

Ganeshkm10, 2022-11-21:
Update: Both gpsd and libgps were analyzed and libgps is chosen for our integration. 
Refer integration issue [here](https://github.com/cu-ecen-aeld/final-project-TejaswiniLakshminarayan/issues/9)

closed 2022-11-21 by Ganeshkm10