cu-ecen-aeld/final-project-TejaswiniLakshminarayan #9: SP2: GPS library integration

prvbharadwaj1731, 2022-11-14:
DoD:

- Succesful integration of library chosen from previous analysis [Refer here](https://github.com/cu-ecen-aeld/final-project-TejaswiniLakshminarayan/issues/8)

Ganeshkm10, 2022-11-20:
Installed gpsd on beaglebone and UART 4 was enabled and connected with GPS sensor.
If we have to design custom uart. Using this deamon would be difficult to integrate. 

[image]

Ganeshkm10, 2022-11-21:
Update: GPS library is integrated and can be found in this [commit id](https://github.com/cu-ecen-aeld/final-project-Ganeshkm10/commit/fb9edc739fff6659e3ba257feb46068f8648e579)

Ganeshkm10, 2022-11-21:
Started with the plan to have a static library for the GPS But while implementing both the library code and the user space code were merged. 

Testing: Tested with Debian distribution. Since UART is still not enabled on the buildroot image. could not test that on our image. However, it is running without any errors. 

[image]

closed 2022-11-21 by Ganeshkm10